node-dmx/dmx #8: Unable to npm install

stevemcquaid, 2015-04-13:
`$ npm install -g dmx`

Results here:
https://gist.github.com/stevemcquaid/45c225fb1be0ea1626bd

npm-debug.log found here:
https://gist.github.com/stevemcquaid/e5bf4241213eae2a7148

wiedi, 2015-04-13:
the node ftdi module requires the ftd2xx headers provided by libftdxx1.

There is some documentation how to install these at <https://github.com/KABA-CCEAC/node-ftdi#prerequisites> and also a script to help with that: https://github.com/KABA-CCEAC/node-ftdi/blob/master/install.sh

If after installing the headers you still have trouble installing feel free to reopen this issue.

closed 2015-04-13 by wiedi

stevemcquaid, 2015-04-13:
Got it! Thank you so much!

One more thing to note - `dmx-web -c` requires a full-path to the conf file, almost got down a rabbit hole trying to get that one solved. 

Ill send a small PR to update docs :)

stevemcquaid, 2015-04-13:
PR here: https://github.com/wiedi/node-dmx/pull/9

Your call, LMK if you want anything changed

wiedi, 2015-04-13:
Great! Looking forward to it :)