node-dmx/dmx #11: enttec-open-usb-dmx fails

deejbee, 2015-10-09:
Hi,  I'm really struggling to get this working on the enttec open usb box.

The first error I get is:

    TypeError: open() expects three agrguments

I've been able to connect to the enttec open usb box by inserting another callback in the open() function in enttec-open-usb-dmx.js:

	self.dev = new FTDI.FtdiDevice(device_id)
	self.dev.open({
		'baudrate': 57600,
		'databits': 8,
		'stopbits': 2,
		'parity': 'none'
	}
	, function(err) {
	    // new callback	
	}
	, function(err) {
		cb(err, device_id)
		if(!err) {
			self.start()
		}
	})
    
Now it seems to connect without error.  My main problem is the light connected to the enttec device doesn't respond to any values.

I have a Stairville LED PAR56 connected and it works correctly when used with Q Light Controller

The DMX Channels for the light are:

    Channel   DMX   Name
    1         1     Mode
    2         2     Red
    3         3     Green
    4         4     Blue
    5         5     Speed control

I edited the dmx-web-example.config so I have a preset which I would expect produce white light:

		{
			label: "White",
			values: {
				'office': { 1:255,  2:255,  3:255  }
			}
		}

The light doesn't go on though.  I don't get what's wrong.  Can anyone help?

wiedi, 2016-04-09:
Hi, the "enttec-open-usb-dmx" seems to be problematic for a few reasons.

The first is that I don't have such a device to test, so I'm always dependant on the community for this driver.
The bigger issue is that it requires very stable timing as the computer generates the DMX messages.

I've just made some changes to the driver that could help with syncing the bus (hopefully) and would be happy if you could test with the current git version. Thanks

wiedi, 2016-10-12:
the enttec-open-usb-dmx driver should finally be fixed with e105013d1ced34f5439edcc41189dd25524fcf1e

closed 2016-10-12 by wiedi

davidhoare, 2016-12-30:
Hi :-)
I've finally got my simply lighting console code to the point where I can test it with an actual device, so I got the entec-open-usb-dmx... of course now I'm having difficulty either connecting or getting anything out of it, so I'm hoping I can work with you to do some testing...
I'm on a macbook air, osx 10.11.6, using the latest git version of your code.
The device shows up as /dev/cu.usbserial-AL00DG7I
It works fine using JV Lightning controller, just as a test, so I know the hardware side works fine...

Starting the app with 
`var universe = dmx.addUniverse('demo', 'enttec-open-usb-dmx', '/dev/cu.usbserial-AL00DG7I');`
yields a quiet (ie. no error log) startup, but I get no response from the device (an LED par), and it indicates it's not receiving any DMX signal.

At this point I'm not certain whether I should be looking at the system or node serial driver, or the specific enttec-open-usb-dmx driver code.

I did a test where I downloaded a fresh git copy, did npm install to get required packages, updated the addUniverse line to specify the "/dev/cu.usbserial-AL00DG7I" device in demo_simple.js, and ran it.
It appeared to function without error (ie. toggled 'on/off' in console) but did not register a proper DMX signal with the light, nor did I get any reaction from it...

I discovered some chatter online about ElCapitan conflicting in some way with FTDI drivers??? That's way outside my knowledge base, but I tried a couple of the suggestions to no avail... I'm not sure if that's a red herring, or in fact what I'm up against and should continue... The fact that the device works perfectly with different software and no system modifications suggests that maybe there isn't a conflict after all, but maybe I'm mis-interpreting...?

Anyway - thanks for a great package, and if you have any ideas that might point me in a direction to test, I'd be happy to be your remote tester :-)
Thanks and happy new year!
David