node-dmx/dmx #21: trouble using with Electron: "DMX is not a constructor"

davidhoare, 2016-11-19:
I'm not sure if this is strictly speaking an issue, or just something I'm having difficulty with in my setup. I've been trying for several days to get this library working in an Electron project I'm testing.

I'm just using the demo_simple.js code as is:

`var DMX = require('./dmx');
var A = DMX.Animation;
var dmx = new DMX();
var universe = dmx.addUniverse('demo', 'null')`

but it bugs out at the constructor, with a `TypeError: DMX is not a constructor`.

Has anyone successfully used this with Electron? Is this error indicative of a simple config problem, or is there some fundamental reason I won't be able to make this work?

Thanks for what looks like a super project,
Cheers,
David Hoare

wiedi, 2016-11-19:
do you have another assignment to the variable `DMX` or a second file called "dmx.js" that will be found instead?

wiedi, 2016-11-19:
if you have installed node-dmx via npm you might want to try `var DMX = require('dmx')`? I'm a bit puzzled how this error can occur.

bluemaex, 2016-11-19:
Hey @davidhoare,

Can you show us a little bit more how you use this library? I also guess there is some naming conflict. 
I just tried using this library inside a super simple electron project and it works for me there.

See the attached zip file for a very simple test based on ``demo-simple`` in electron: [test-electron.zip](https://github.com/wiedi/node-dmx/files/601792/test-electron.zip) - does this work for you?

--maex

davidhoare, 2016-11-20:
Hello :-) Thank you both for responding!
Re the first error - - you were right: there was another file in the same folder (dmx.json) that I was using as a config file... 
However, now that I've cleaned that up, I'm getting a version conflict error with both my project and the sample that Max provided. 
`Module version mismatch. Expected 49, got 48.`
I think I can trace this to the serialport module (maybe :-) because it was what I had to install to the local node_modules folder in Max's sample in order to get it to launch. Now that it launches, I get that version mismatch error in console.
What version of node are you using or _should_ this be compatible with. I cannot find version '49' anywhere - - it goes from 48-->51 in terms of official releases... (https://nodejs.org/en/download/releases/)

Thank you for your help! I think I'm getting closer :-)
DH

bluemaex, 2016-11-20:
I think a ``npm install electron-rebuild`` and then ``./node_modules/.bin/electron-rebuild  -$(electron -v)`` should fix that issue.

davidhoare, 2016-11-20:
god bless you! I've been fighting with this for days!!!
I'm (obviously :-) new to Electron, and I thought it would be a fun way to create a little cross-platform version of a lighting console program I did years ago in VB... Little did I imagine the headaches getting it setup properly. I hope this is it now (appears to be working just fine) so thank you again ever so much for your time tonight.
Cheers!
David

wiedi, 2016-11-20:
glad it's working!
Have fun :)

closed 2016-11-20 by davidhoare

bluemaex, 2016-11-20:
Woho, good news! 🎉 
Have fun playing around with it 👍

nucleardreamer, 2016-11-20:
@davidhoare another thing which you might want to do (unless you have a specific other need) is make sure you are using the same version of node as electron is built with. Right now, if you are on latest electron, you should be using 6.5.0, the [release page](https://github.com/electron/electron/releases/tag/v1.3.7) is a good place to check often =)