node-dmx/dmx #24: entec-open-usb-dmx not responding?

davidhoare, 2017-01-01:
Hi :-)
I've finally got my simply lighting console code to the point where I can test it with an actual device, so I got the entec-open-usb-dmx... of course now I'm having difficulty either connecting or getting anything out of it, so I'm hoping I can work with you to do some testing...
I'm on a macbook air, osx 10.11.6, using the latest git version of your code.
The device shows up as /dev/cu.usbserial-AL00DG7I
It works fine using JV Lightning controller, just as a test, so I know the hardware side works fine...

Starting the app with 
`var universe = dmx.addUniverse('demo', 'enttec-open-usb-dmx', '/dev/cu.usbserial-AL00DG7I');`
yields a quiet (ie. no error log) startup, but I get no response from the device (an LED par), and it indicates it's not receiving any DMX signal.

At this point I'm not certain whether I should be looking at the system or node serial driver, or the specific enttec-open-usb-dmx driver code.

I did a test where I downloaded a fresh git copy, did npm install to get required packages, updated the addUniverse line to specify the "/dev/cu.usbserial-AL00DG7I" device in demo_simple.js, and ran it.
It appeared to function without error (ie. toggled 'on/off' in console) but did not register a proper DMX signal with the light, nor did I get any reaction from it...

I discovered some chatter online about ElCapitan conflicting in some way with FTDI drivers??? That's way outside my knowledge base, but I tried a couple of the suggestions to no avail... I'm not sure if that's a red herring, or in fact what I'm up against and should continue... The fact that the device works perfectly with different software and no system modifications suggests that maybe there isn't a conflict after all, but maybe I'm mis-interpreting...?

I also note that deejbee above is using totally different code to init the device:
`self.dev = new FTDI.FtdiDevice(device_id)`
and I'm not sure if perhaps I should be trying something like that as well...?

Anyway - thanks for a great package, and if you have any ideas that might point me in a direction to test, I'd be happy to be your remote tester :-)
Thanks and happy new year!
David

davidhoare, 2017-01-01:
just to share further troubleshooting...
I've created a simple test file using only the serialport module:

`
var SerialPort = require('serialport');
SerialPort.list(function (err, ports) {
  ports.forEach(function(port) {
    console.log(port.comName);
    console.log(port.pnpId);
    console.log(port.manufacturer);
  });
});

var port = new SerialPort('/dev/cu.usbserial-AL00DG7I');
port.on('open', function() {
  port.write('main screen turn on', function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  });
});	

// open errors will be emitted as an error event 
port.on('error', function(err) {
  console.log('Error: ', err.message);
})
`
...and it seems to find the device and successfully open and write to it... 
don't know where to go from here, but I'll keep plugging away at it :-)
Thanks!
David

wiedi, 2017-01-15:
Yes, sadly the entec-open-usb-dmx driver is still somewhat problematic.
Sometimes it works, sometimes it does not. If I send smaller/shorter universes the chances of it working improve, so I suspect something is still wrong with the timings or the way to device is accessed. Tweaking with the interval also makes it work better or worse depending on the connected lights. I've not yet found a reliable way to make it work "always".

One thing I don't see in your test code is configuring the baudrate. Not sure if the defaults are correct?

seriousManual, 2017-06-29:
hey @wiedi 
I'm currently struggling with the OpenDMX USB too. 
I've tried several approaches but was not able to get anything out of the device, when using qlc+ it works perfectly fine.

above you wrote that under certain conditions you were able to get it to work, can you say which configuration you made exactly?
maybe i can reproduce that behaviour...

thank you!

wiedi, 2017-06-29:
Not exact conditions as it always was a bit unreliable, but as noted shorter universes helped. Tuning the interval also had an influence, but I could not determine a value that was independent of the lights on the bus, so you'll have to experiment with your setup.

seriousManual, 2017-06-30:
i saw that you wrote earlier about the shorter universes (you mean like shortening the Buffer from 512 to lets say 200 right?), I tried that too, to no avail.
also trying different intervals did not help.

wiedi, 2017-07-01:
yes, but much shorter, like 16 to 32.
And it also dependet very much on the lights, some reacting very different from others.

seriousManual, 2017-07-02:
i actually gave up on the Open DMX USB and bought an Enttec USB Pro instead...
that worked fine out of the box. :)

dreammonkey, 2018-02-06:
hi people,
sorry for intruding, but I'm currently trying to talk to an eurolite usb-dmx512-pro (mk2) from mac osx (sierra 10.12.6)... Needless to say I'm struggling... It would be extremely helpful if someone could give me some pointers in the right direction... 

I've managed to talk to it using Lightkey.app
-> it recognises the devices on startup but it always prompts me for my admin (sudo) password, which I find to be weird... (but at least I was hopeful the the light was responding to my input)

I've tried using ola, but I'm not certain in which plugin I should configure the device (I'm guessing open dmx or am I mistaking?)

I've not yet successfully operated the device using node-dmx, pretty much for the same reason: I don't know where to put it in my config file... 

FYI: I'm trying to address 1 universe, 1 spot, a showtec Kanjo60 (10channels)

I installed these drivers:
D2XX: /usr/local/lib/libftd2xx.1.4.4.dylib
/System/Library/Extensions/euroliteusbshield.kext

Any advice is much appreciated !

dreammonkey, 2018-02-07:
OK, to answer my own question: the eurolite usb-dmx512-pro mk2 works fine using enttec-usb-dmx-pro driver !
Not sure which/any of the (mac) drivers had anything to do with it though...

Fensterbank, 2018-02-08:
@dreammonkey Did I understadt in correctly that you had problems controling the eurolite usb-dmx512-pro mk2 with entec-open-usb-dmx but it worked without struggle using the enttec-usb-dmx-pro?

Did you only change the selected driver in your config or did you change the entire USB-DMX-Adapter?

kfstudio, 2018-02-19:
Hi,

I have same problem, so no signal go out of device. It's a soft issue but we can resolve by hardware.
The rts pin is not controlled well (stay low in my case), so i cut between the rts of the FT232RL (pin 3) and the pin 3 of the 75176B, and connect the pin 3 to the pin 8 of the 75176B.
That solve my problem.

There is also a bug in config of serial port, but i do a pull request for that !

wiedi, 2018-02-19:
Wow, thanks for tracking this down! This sounds very promising.

wiedi, 2018-02-19:
Indeed that was the issue. I'm happy to report that with f27a9c9ea08d648c823ae616bf8ab1941278eb8a the `entec-open-usb-dmx` is finally somewhat usable.

In my testing there are still sometimes tiny issues, but overall this seems to work pretty ok now.

closed 2018-02-19 by wiedi