node-dmx/dmx #25: Not working on node version 7.x.x

willcook4, 2017-03-09:
I'm running this in node 6.9.4 and I get the following in the terminal on running the script.

```
... Output above...

==== C stack trace ===============================

 1: v8::Template::Set(v8::Local<v8::Name>, v8::Local<v8::Data>, v8::PropertyAttribute)
 2: SerialportPoller::Init(v8::Local<v8::Object>)
 3: init
 4: node::DLOpen(v8::FunctionCallbackInfo<v8::Value> const&)
 5: v8::internal::FunctionCallbackArguments::Call(void (*)(v8::FunctionCallbackInfo<v8::Value> const&))
 6: v8::internal::MaybeHandle<v8::internal::Object> v8::internal::(anonymous namespace)::HandleApiCallHelper<false>(v8::internal::Isolate*, v8::internal::(anonymous namespace)::BuiltinArguments<(v8::internal::BuiltinExtraArguments)1>)
 7: v8::internal::Builtin_HandleApiCall(int, v8::internal::Object**, v8::internal::Isolate*)
 8: 0x3310d7e092a7
 9: 0x3310d7f77244
(node) v8::ObjectTemplate::Set() with non-primitive values is deprecated
(node) and will stop working in the next major release.

==== JS stack trace =========================================
... Output below ...

```

I'm guessing that there is some piece of a dependency that node-dmx uses `v8::ObjectTemplate::Set()`. How can I help find the corresponding dependency and see if there is an update to work in node v7.x.x?

wiedi, 2017-03-09:
Thank you for the report. I don't have an immediate answer.

Maybe the js stack trace includes helpful information?
You could also try to just blindly bump dependency version numbers as there aren't too many (serialport would be my prime suspect).

wiedi, 2017-03-09:
Oh yes, ```2: SerialportPoller::Init(v8::Local<v8::Object>)``` actually hints to node-serialport.

willcook4, 2017-03-10:
You were right. It appears that serialport@4.0.7 is the latest package in the npm repos. I tried with `"serialport": "latest"`. According to their GitHub repo page, they are working on v5 now which is in beta. This appears to have node V7 support.

Running `npm outdated` gave me the following results from the node-dmx package.json :  

```
Package         Current  Wanted  Latest 
commander        2.0.0   2.0.0    2.9.0 
connect          2.11.2  2.11.2   3.6.0 
express          3.4.8   3.4.8    4.15.2  
socket.io        0.9.17  0.9.17   1.7.3
```

I would love to do some testing and do a Pull Request for you but I'm out of time at the moment.
Just thought I would let you know what I had found. I'm a fan of node-dmx and keen to see it develop.

willcook4, 2017-03-13:
Just had an update from the team over at serialport. I reported to them that version 4 is the only version I saw available in the npm reops. Now they have updated and `npm install serialport@5` will get version 5. I have not tried this yet but hope to get a chance in the near future to try it out with node-dmx and then node v7.x.x. I'll let everyone know here how I get on or if anyone gets there before me, post your results here.

jmbajo, 2017-04-10:
Hi, I had the same problem recently, I'm using Node 6.10.2.

I solved it using the git version of the library and not the one listed in the 'npm' repos (which is older).
To do this you have to use npm install without arguments in a directory which contains an updated node-dmx package.json file (for example, the one in this git repo).

westhom, 2017-06-21:
If anyone is interested in a temp fix, I've forked node-dmx and updated the serialport dependency ref to 5.0.0-beta. Also made a simple change to `enttec-usb-dmx-pro.js` to reflect a recent API change in serialport. I've tested on node 7.x (MacOS and Raspberry Pi) with my Enttec USB Pro MK2 and it seems to work.

`npm install westhom/node-dmx`

wiedi, 2017-06-21:
The state of backwards compatibility with node and node-serialport is somewhat depressing.
For what it's worth i've bumped the version and uploaded a current one to npm.

I'm not so sure about adding a beta version as a dependency. Especially as it seems to drop support for everything before node 8.

westhom, 2017-06-22:
@wiedi Agreed. I don't think you should add a beta dependency, especially considering a 5.0 release may happen soon. I just needed to get the lib working in a short amount of time for a project and wanted to report back the results of my testing w/ a new version of serialport.

alexanderwallin, 2017-06-26:
@westhom You totally saved my week with that fork. Thanks for publishing!

closed 2018-01-10 by bluemaex

bluemaex, 2018-01-10:
Just released 0.0.6 by a small delay - which now has updated dependencies and works on node 8 and 9 again 🎉

PR #37 (2018-01-10): Update dependencies for node 8
Updated all the dependencies to their current versions as of today. I have this new setup running successfully on Node 8.9.4

Merging this pull request fixes #25 and make #31, #32, #33 obsolete.