node-dmx/dmx #36: Documentation is wonky for this project

videohead, 2018-01-06:
No offense to the awesome work done here to make this functional, but I found the documentation and setup for this to be very wonky and not helpful.
Using DMXUSB Pro here is what I got to work:
var universe = dmx.addUniverse('demo', 'enttec-usb-dmx-pro', 'COM3') //on Windows find Com port in device manager
Then I found that the JSON values to be really simple, it's just channel:value,
The documentation makes it seem complicated, and the examples are not commented at all.
So you need to look in the manual for your fixture and get the channel numbers for the functions, eg. channel 1 for intensity. You also need to set your fixtures correctly with the right starting channel.
BUT!
Each channel value is mysteriously decremented by 1 value. So if you set your fixture to start at 1, you start your values with 0.
Confused? So was I. My gobo wheel spinning and spinning.
Anyway, got it to work fine with a DMXPro USB from Enntec .
If you have questions you can DM me.
I will post more when I get it working in UI sliders. ;)
Here was something that worked with the Chauvet intimidator 155. 
setInterval(function(){
  universe.update({ //for some weird ass reason each channel in the map is decrement by 1?
  		0:50, //so this is pan
  		1: 100, // tilt
  		5:12, //sets color to magenta
  		6:255, //shutter open
  		7:255, //intensity full
  		8:127, //gobo wheel off
  		}); 
  });
What was also helpful was having another file where all values were set to 0 = easy reset.

Fensterbank, 2018-01-06:
> Each channel value is mysteriously decremented by 1 value. So if you set your fixture to start at 1, you start your values with 0.
I also found this out the hard way last week. The channel index is starting with 0.

Yes, the project needs more documentation and maintenance. This surprises me, as it is obviously the only DMX-control based on NodeJS. But in summary it's working, and that's more important than a complete documentation.

wiedi, 2018-01-08:
Sorry about not staying on top of things.

I'd be happy about PRs to improve the Readme and additional documentation.

The issue with channel offsets is that DMX Documentations start at offset 1 and node.js Buffers at offset 0. This should be addressed by 00d960e441024627e1023e2ac0776b899ed6b9af which will be in the next tagged release.

closed 2018-01-08 by wiedi