node-dmx/dmx #40: Broadcast on artnet

Joebayld, 2018-01-11:
I need to enable broadcast on the artnet driver. I added this code to make it work. 

```
self.dev.on('listening', function(){
    self.dev.setBroadcast(true);
});
```

Should we add it to the master? Or maybe have an option to enable broadcast?

wiedi, 2018-01-11:
The setBroadcast() call seems reasonable.
Is the event handler for 'listening' actually required or could we just do setBroadcast directly below the `self.dev = dgram.createSocket('udp4')` line?

Joebayld, 2018-01-11:
I tried adding it below and it didn’t work. Upon research, I found this solution and it works. 

Joe

wiedi, 2018-01-11:
ok great, thanks for confirming that. Do you want to create a pull request?

wiedi, 2018-01-11:
merged, thanks!

closed 2018-01-11 by wiedi