node-dmx/dmx #42: Enhancement: Allow managing fixtures/scenes in the web interface

Joebayld, 2018-01-11:
I think it'd be a good idea to edit scenes / fixtures in the web UI. Perhaps under a 'config' tab.

I can start working on this however how would I modify the universe? I would have to save to the json config, but it would need to change the current DMX output in realtime as well. Perhaps a step in the right direction and I can work on this. 

Thoughts?

Fensterbank, 2018-01-12:
> I can start working on this however how would I modify the universe? I would have to save to the json config, but it would need to change the current DMX output in realtime as well. Perhaps a step in the right direction and I can work on this.

You would either use websockets or the HTTP methods (take a look at `dmx-web.json`).

> I think it'd be a good idea to edit scenes / fixtures in the web UI. Perhaps under a 'config' tab.

Personally, I'm not sure about it. The web interface is very rudimentary and not a good code base for further development (e.g. scripts and styles inside the `index.html`). Improving the existing web interface is a bunch of work (at the moment it doesn't even support animations) and the point is, that node-dmx is mainly used as a library, e.g. by installing the npm package. (take a look at https://github.com/wiedi/node-dmx/network/dependents to see wich projects on Github use it)

So technically we should avoid creating a huge web app with many files in this package, because that's things 90% of the people don't want in the package.

In general, I have the same opinion as you: It would be good, if we have a modern web interface where we can create scenes, fixtures, even animations. But I think it should be done in another project (e.g. as a modern react application) which uses either `node-dmx` as library or `node-dmx` as webserver.

But as I'm not the founder of this project, I'm interested what @wiedi thinks about this?

mtraeger, 2018-01-12:
I started some time ago with enhancing the web interface with color animations, fading, simple beat detection and other improvements to bring it to a collaborative lightning control (find current state at https://github.com/mtraeger/dmx-webcontrol - not in the list as currently not depending on the library). 

I originally expanded the dmx-web.js and index.html for trying out some things. And @Fensterbank is right: this way it is quite a mess right now, especially in the index.html. Fyi: Im planning to rewrite the web interface with react or vue and extract this dmx library. 

I mailed with @wiedi over a year ago what he thinks about including my improvements in this project, but he preferred to keep this library simple and maintainable. 

Currently, I would also suggest keeping this library clean and simple.

Joebayld, 2018-01-14:
Got it. Well then I like the direction you're headed. A different web library that references this one would be nice. Would just like to be able to collaborate on the web portion.. 

My main goal is to be able to add/remove fixtures without having to modify the JSON and relaunching.

Thanks for the input though.

closed 2018-01-14 by Joebayld