node-dmx/dmx #56: How to configure the driver on windows?

fralbo, 2018-07-02:
hello,

Looking at the config file, I'm wondering how to specify the driver access point on Win 10?

BR,
Alain

bluemaex, 2018-07-02:
@fralbo Sorry I can't really help you here as I do not have a  Windows machine to test on, but as far as I remember the SerialPort are called something like COM2, COM3 etc.

We use the nodejs package ``serialport`` which has support for windows and I found this gist which might help you testing it: https://gist.github.com/maximilian-lindsey/b5a981d3657627bfbb89
If you suceed please let us know, then we update the documentation!

Cheers
Max

fralbo, 2018-07-03:
Yes, I suppose it should be as simple. 
I'm waiting for the device. I'll let you know ASAP.

apsylone, 2018-11-06:
@fralbo I've you found a solution to this ? :-) 

Thanks,

bluemaex, 2019-01-20:
I guess this issue got resolved due to no response in the last 4 months - Feel free to reopen it if there are still questions!

closed 2019-01-20 by bluemaex

sebbean, 2019-12-02:
what's the solution?

apsylone, 2019-12-03:
@sebbean try to use COM1, COM2... as @bluemaex mentionned :-)

gabrielstuff, 2020-07-13:
Hello, 

I'm starting to try `dmx-web` on windows. So far, I'm happy. You can check here https://gist.github.com/gabrielstuff/bb3b69926c32cf5164467b1a45fc7c47 how to do. Basically, you look at your serial settings and find what is the `COM` : `COM1`, `COM2`, `COM3`, `COM4`, `COM5`

It should show as `Serial USB (COMX)`