node-dmx/dmx #65: No connection to Enttec Open Dmx

YannickHaas, 2018-09-06:
What is the correct path to my enttec device? 
pi@raspberrypi:~ $ dmx-web
**Error: Error: No such file or directory, cannot open /dev/cu.usbserial-6AVNHXS8**
^C
pi@raspberrypi:~ $ cd /sys/bus/usb/devices
pi@raspberrypi:/sys/bus/usb/devices $ ls
1-0:1.0  1-1  1-1.1  1-1:1.0  1-1.1.1  1-1.1:1.0  1-1.1.1:1.0  1-1.1.2  1-1.1.2:1.0  1-1.1.2:1.1  1-1.1.2:1.2  1-1.2  1-1.2:1.0  1-1.3  1-1.3:1.0  1-1.3:1.1  1-1.3:1.2  usb1
pi@raspberrypi:/sys/bus/usb/devices $ cd
pi@raspberrypi:~ $ sudo udevadm monitor
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[57347.509619] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2 (usb)
KERNEL[57347.517134] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0 (usb)
KERNEL[57347.517294] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/ttyUSB0 (usb-serial)
KERNEL[57347.518289] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/ttyUSB0/tty/ttyUSB0 (tty)
UDEV  [57347.539482] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2 (usb)
UDEV  [57347.548009] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0 (usb)
UDEV  [57347.553368] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/ttyUSB0 (usb-serial)
UDEV  [57347.575536] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/ttyUSB0/tty/ttyUSB0 (tty)

wiedi, 2018-11-14:
@YannickHaas try `/dev/ttyUSB0`

bluemaex, 2019-01-20:
I guess this issue got resolved due to no response in the last 4 months - Feel free to reopen it if there are still questions!

closed 2019-01-20 by bluemaex

wiedi, 2019-01-20:
There is also an utility included with the node-serialport package to list devices that might help others in the future: https://serialport.io/docs/en/guide-cli