node-dmx/dmx #80: EnttecUSBDMXPRO driver: SerialPort "writable" property no longer exists?

tracycollins, 2019-03-30:
In the sendUniverse method, it checks the "writable" property:
```
EnttecUSBDMXPRO.prototype.sendUniverse = function () {
  if (!this.dev.writable) {
    return;
  }
```
I think that it should be checking SerialPort.isOpen property instead? I can't find the writable property in the current SerialPort documentation: [https://serialport.io/docs/en/api-stream](https://serialport.io/docs/en/api-stream)

I found this while debugging a memory leak when testing an animation to be sure my app wouldn't crash if it couldn't find an Enttec device. If testing for 'writable', the driver will write to a non-existent serial port, leak memory, and eventually crash. When I changed the driver code to this, it seems to run ok:

```
EnttecUSBDMXPRO.prototype.sendUniverse = function () {
  if (!this.dev.isOpen) {
    return;
  }

```

Fensterbank, 2019-11-06:
Good catch, in the [documentation of the serial port version 7.x.x](https://serialport.io/docs/7.x.x/api-stream) we are using there is also no mention of `writable`.
`isOpen` exists since serialport 5.0.0 while this code was once created using serialport 4, so someone should make a pull request and fix this.

But I'm wondering: if `writable` would be `undefined`, the driver would not work at all... Is `writable` still implemented in the serialport codebase or how does this even work? o_O

Granjow, 2021-03-25:
It is coming from the [stream writable](https://nodejs.org/api/stream.html#stream_class_stream_writable) and still exists :)