node-dmx/dmx #94: module not working on macOS Catalina?

davidsi, 2020-01-07:
I have a enttec open usb dmx and a ultra dmx micro and I can not get either to work on my MacBook Pro. (Connected via a dongle as the MacBook is all usb C). I modified the simple demo, as below, and neither work. (I comment/uncomment and plug/unplug the correct device pair) 

I'm connecting the device(s) to a dmx512 decoder for outputting to led strips. The strips definitely work as I can connect them to power directly. The /dev/usbserial addresses are correct.

Any help appreciated.

const DMX = require('./index');
  
const dmx = new DMX();

  const universe = dmx.addUniverse( 'demo', 'enttec-usb-dmx-pro',  '/dev/cu.usbserial-6A009910' );
//const universe = dmx.addUniverse( 'demo', 'enttec-open-usb-dmx', '/dev/cu.usbserial-AL05O8DK' );

let on = false;

setInterval(() => {
  if (on) {
    on = false;
    universe.updateAll(0);
    console.log('off');
  } else {
    on = true;
    universe.updateAll(250);
    console.log('on');
  }
}, 1000);

gloverab, 2020-08-20:
Did you ever get any resolve on this? I believe I'm having the same issue.

davidsi, 2020-08-20:
No :-(

David