node-dmx/dmx #95: Search for devices

hrueger, 2020-02-07:
The library is working great. A nice-to-have feature would be the ability to search for connected devices. Happy to implement that, but what would be the best approach? Should we add a `search` function to each driver and call them one after each other / simultaneously?

wiedi, 2020-02-07:
Sounds good.

The [RDM Spec](https://en.wikipedia.org/wiki/RDM_(lighting)) calls this functionality "discovery", so maybe `discover()` instead of `search()`?

I'm unsure how much hardware actually supports this. I know the Enttec DMX Pro does but many other devices probaby don't.

hrueger, 2020-02-07:
I'm sorry, I was talking about searching for interfaces, not the actual lights...

wiedi, 2020-02-07:
I see.

That's going to be hard on the per driver level. Most would just report the same serial ports.
I also suspect that filtering by USB vendor IDs is going to be very unreliable.

The serialport package has a [serialport-list](https://serialport.io/docs/guide-cli) command - is that something similar to what you had in mind?

hrueger, 2020-02-07:
> The serialport package has a serialport-list command - is that something similar to what you had in mind?
Yes. I'll look into that.

hrueger, 2020-02-11:
So my original idea was to find a connected device using `serialport-list`. But then we would need to find out what driver to use, and as far as I found out, this would be pretty hard. The beaglebone interface for example can't respond to incoming data, so we could not check if it is that one.
In addition we would leave artnet devices out.

Any more thoughts on that?

wiedi, 2020-02-23:
Yeah I'm not sure we can create a reliable way to discover all possible devices and make correct driver suggestions. :/

Maybe the best we can do is add some documentation about `serialport-list` that helps people figure out their device path. There aren't many driver options and they probably know what they physically plugged in so that maybe a driver suggestion isn't needed anyway.

For artnet other stuff on the networking side needs to be configured as well so I'd hope people setting this up know what they are doing. Maybe more documentation could also help here - like a guide on the wiki or something.