node-dmx/dmx #99: Add types for typescript support

nofalx, 2020-03-17:
Hi 
im using your library for a project and it seems to lack types for typescript. I believe adding support for it would really make this library more awesome. I can help you with this and add types. Would you be willing to add support if I open a PR? Thank you

bluemaex, 2020-03-18:
@nofalx Yes, please feel free to add a PR for types, I would happily review it!

Granjow, 2020-10-31:
@bluemaex In case I get this working for the Starville LED Pixel Rail – would you accept a MR which migrates this project to TypeScript?

bluemaex, 2020-11-01:
@Granjow I would be totally fine with that! But please in that case please also add a build target and instructions :)

Bordman-ger, 2020-11-09:
@Granjow  Maybe I can help you? I am getting the same problem (trying to get an IOBroker adapter working using TypeScript) maybe we can work on this together? or at least I can help testing ....

Granjow, 2020-12-16:
Sorry for the late response. Ended up using ArtNet for the LED Rail (https://www.npmjs.com/package/artnet-base). We have another DMX device though and I will probably need the library there. If I do, I will do the TypeScript port (should be fairly straightforward), testing would definitely be very helpful then!

closed 2021-10-10 by wiedi