node-dmx/dmx #148: EnttecOpenUSBDMXDriver

ScreamZ, 2023-01-13:
hello,

I'm using it with a 8-channel mode with that.
[CLNB600_Cameo_DMX_Control_Table_EN.pdf](https://github.com/node-dmx/dmx/files/10412656/CLNB600_Cameo_DMX_Control_Table_EN.pdf)

I'm just trying to move to proper position with a white light

```ts
import { DMX, EnttecOpenUSBDMXDriver, Animation } from "dmx-ts";
import { EnttecOpenDMXUSBDevice } from "enttec-open-dmx-usb";

const dmx = new DMX();

enum C {
  MechPan = 1,
  MechTilt = 2,
  Dimmer = 3,
  Strobe = 4,
  Red = 5,
  Green = 6,
  Blue = 7,
  White = 8,
}

(async () => {
  const universe = await dmx.addUniverse(
    "demo",
    new EnttecOpenUSBDMXDriver(await EnttecOpenDMXUSBDevice.getFirstAvailableDevice()),
  );
  universe.updateAll(0);
  universe.update({ [C.MechPan]: 255, [C.Dimmer]: 255, [C.White]: 255 });
})();

```

But this is glitchy as hell look at this :

https://user-images.githubusercontent.com/6640835/212352663-b77a8078-423d-4cf9-a5be-3a5a5244f123.MOV

What I'm supposed to understand ?

hrueger, 2023-01-13:
Hi @ScreamZ,
the readme states the following:
```
enttec-open-usb-dmx: driver for "Enttec Open DMX USB". This device is NOT recommended, there are known hardware limitations and this driver is not very stable. (If possible better obtain a device with the "pro" chip)
```
I cannot watch the video, I only hear audio. Maybe it only works on Mac? Don't know.

However, you might want to try the following things:
- Play around with the [options](https://github.com/node-dmx/dmx/blob/faf42e00b2904c2e3b8dced190de8a65733b6029/drivers/enttec-open-usb-dmx.js#L11-L14). Maybe lower the baud rate?
- You could also try the TypeScript Version of this library, https://github.com/node-dmx/dmx-ts It has slightly more up-to-date drivers.
I don't think that this is going to change much, though, as your code seems to send the values only once. That means, flickering may be introduced by the Interface and not the library?

Are you using a DMX Terminator?

ScreamZ, 2023-01-13:
I changes the video to mp4, check it :)

For now I've this one, I'll check if  ifound something but this gets expensive and i want to be sure its related to interface.

I'm using the typescript already, really nice.

Yes I'm using a terminator. Its the only device. The baud rate seems to be a constant of the protocol, ima check,.

Also used : https://github.com/moritzruth/node-enttec-open-dmx-usb/issues/82 but have other issues.

I'll maybe try implement a new driver ?

moritzruth, 2023-01-13:
I also initially tried this library but because it didn’t work I wrote https://github.com/moritzruth/node-enttec-open-dmx-usb.

I assume it has to do something with the timing of `BRK` and [slight inaccuracies of `setTimeout`](https://nodejs.org/api/timers.html#settimeoutcallback-delay-args), but I really don’t feel like debugging that stuff. Especially as it is working for me.